mozilla/project_haiku.iot #10: Battery power the device

punamdahiya, 2016-05-02:
Investigate hardware options to keep the device powered for 4-5 days
a) Using power shield to recharge LIPO battery
b) Device in sleep state when not connected to WiFi

sfoster, 2016-05-02:
I have a couple of battery mgmt and charging board getting shipped to me to work on this

sfoster, 2016-05-09:
In the Bill of Materials in the README (commit 5f06df7) I've included the Adafruit LiIon/LiPoly charger modules and a LiPoly 2000mAh battery. This lets us recharge the unit by plugging it into a USB wall adapter or computer etc. (The cable for USB-A to USB-MiniB is also in the BOM)

Initial testing shows the 2000mAh battery can power the device for 12+ hours, we'll need to do more testing when we've got more devices assembled and ready for use. 

Note that we are simply wiring together the Photon to the charger module to the LiPoly battery itself. Each is an off-the-shelf component and my understanding is that we should inherit their guarantees and certifications for safety, compliance etc.

closed 2016-05-09 by sfoster