mozilla/project_haiku.iot #13: Solution for logging activity between 2 paired devices

sfoster, 2016-05-02:
Implement and/or document how we'll logging of activity between 2 paired widgets over the course of the study (or some time period). We may be able to use Particle's dashboard for this, or export data and create our own report format that displays the timeline

punamdahiya, 2016-05-04:
curl https://api.particle.io/v1/devices/events?access_token=<accesstoken> returns data in json format once the device is online

punamdahiya, 2016-05-04:
One option is to look into possibility of capturing and persisting json data with the spreadsheets available in Google Drive

kglazko, 2016-05-06:
Alright! Here goes my first implementation :)

##Web IDE and Code##

- Log into the ParticleIO Web IDE and upload your existing script. I am using a button flashing script. 
- In my case, I created an event called LED with the following code: `Particle.publish("LED","on",16777215,PUBLIC);`
- You can create and publish as many events as you want, and name them anything that you want.

##Zapier##

- Head over to the Zapier website and create an account. Don't worry, it's very quick!
- Choose to make a zap that involves Webhooks and Google Drive.
- Follow the zap instructions, and when it comes to copy/pasting the Webhooks URL, copy the URL and continue to the next step

##Integration##

- Log into https://dashboard.particle.io and click on the 'Logs' icon. Make sure your particle shows up there.
- Click on the log. If the code is executing, you should see the events you created being displayed there. If not, make sure your code is running during the next step.
- Go to the Integrations tab and click on the Webhooks option. Name the event with the exact same way one of your events is named (case matters!), and paste in the webhooks URL from Zapier you got earlier. Save the webhook.

##Zapier, Cont'd.##

- After the copy/paste step, you will be asked to test the webhook. Do this, then continue through the instructions and save the Zap.
- Activate the zap.

##On your dashboard##

- You will be able to see POST requests being sent to the web hook in the Log section.

##On your Google Spreadsheet##

- You will be able to see the info that you designated to be sent during the Zapier set-up being sent to the spreadsheet. This can include JSON child info, etc.

kglazko, 2016-05-06:
All this coolness was sent via Webhooks :) https://docs.google.com/spreadsheets/d/1wWXVHtKRUui7LPdpyGS1LMUz9TQ-Mc5zG08bPvZhOdo/edit#gid=0

sfoster, 2016-05-26:
We have a solution here that is working for the UR#2. We'll revisit in a new issue for future improvements in further studies.

closed 2016-05-26 by sfoster