mozilla/project_haiku.iot #15: Implement pairing of 2 devices

sfoster, 2016-05-02:
We need a way for each device to listen for and publish events to a particular device which is its "pair". This can be done on the device e.g. some constant we populate in the code before flashing to the device, or the pairing could be done by the message "broker" which knows which device to relay messages to.

punamdahiya, 2016-05-06:
Another alternative to get 5 pairs ready is
1. Create a particle login [https://dashboard.particle.io/login] and claim the two buttons that should be paired under same login

2. This will ensure private publish and subscribe messaging between paired devices and their logs can be seen under single particle login

3. Flash Device 1 with code https://github.com/mozilla/smarthome.iot/tree/master/Prototype-UR2/InternetButton/buzzy-pixel 

4. Update ButtonConfig.h by uncommenting line 14 and 15 and commenting line 8 and 9. Flash second device

Devices should be paired and communicate when connected to wifi network

punamdahiya, 2016-05-26:
We have pairing implemented and the details documented in user spreadsheet. Closing the issue.

closed 2016-05-26 by punamdahiya