mozilla/project_haiku.iot #4: Implement "connecting" LED animation

sfoster, 2016-05-02:
Write code (or using existing code) to display a recognizable color/blink sequence to communicate to the user that the device is attempting to connect to a network

punamdahiya, 2016-05-06:
For the scope of this prototype we can leverage LED status available on photon. If photon is blinking green, it is trying to connect to the internet. 
If Wi-Fi credentials are available to photon, in few seconds device connects and starts breathing cyan.

sfoster, 2016-05-09:
When we document wifi setup, lets add a note about the status LED. At that point we can consider this issue closed

punamdahiya, 2016-05-26:
Documented in help guide handed to research participants, closing the issue.

closed 2016-05-26 by punamdahiya