mozilla/project_haiku.iot #19: Reset mechanism

sfoster, 2016-05-07:
We need to make sure we have a simple way to reset the device, and wipe any stored wifi credentials. We may ask study participants to reset when handing back the devices just so there's no possibility those credentials could hang around and be used/abused after the study. 

Possibly the 'reset' button on the Photon already does this. Lets confirm before resolving this issue.

kglazko, 2016-05-09:
I will investigate this.

kglazko, 2016-05-09:
Requires a Factory Reset- this can be done via the Mode button

Procedure:

The procedure is same as the one described above (DFU Mode), but in this case you should continue holding down the MODE button until you see the Core change from blinking yellow to blinking white. Then release the button. The Core should begin after the factory reset is complete.

   1. Hold down BOTH buttons
   2. Release only the RST button, while holding down the MODE button.
   3. Wait for the LED to start blinking yellow (continue to hold the MODE button)
   4. The LED will turn solid white (continue to hold the MODE button)
   5. Finally, the LED will turn blink white rapidly
   6. Release the MODE button

 https://docs.particle.io/guide/getting-started/modes/core/

punamdahiya, 2016-05-10:
I believe factory reset will wipe the code flashed on the device. If our goal is to wipe Wifi credentials stored on device, can we check if have a way to only delete stored WiFi without doing factory reset

punamdahiya, 2016-05-10:
Link here shows how to erase stored wifi network,  https://docs.particle.io/guide/getting-started/modes/core/#wi-fi-network-reset . We should test if this will suffice our need

punamdahiya, 2016-05-26:
Updated https://github.com/mozilla/smarthome.iot/tree/master/Prototype-UR2 with reset instructions. Marking issue closed.

closed 2016-05-26 by punamdahiya