mozilla/project_haiku.iot #20: Add open source license to the repo

sfoster, 2016-05-09:
We want a license for the overall repo - which should be compatible with any open source licensed "libraries" and other software that we drew upon to devise "our" software that we are placing in GitHub. I.e. compatible with the LGPL license used by Particle for their firmware: https://github.com/spark/firmware/blob/develop/LICENSE

ezoehunt, 2016-05-10:
Quick google search turned up this for how to choose a license for:
[Open Source Hardware](http://www.open-electronics.org/how-to-choose-your-open-source-hardware-license/)

Assuming open source hardware license is new to Mozilla. Shall we ping Braden about this?

sfoster, 2016-05-10:
For now I've added an MIT license to the repo: 6d7c9e2 which as I understand it is compatible with the LGPL license Particle have attached to their firmware (though we are not re-distributing the firmware in our repo, we are building on top of it)

We'll need an appropriate license for the 3d files, also the overall design which we'll document (parts list, wiring diagram, final assembly). I have a recollection of classic software licenses like MIT being unsuited for these kind of assets?

ezoehunt, 2016-05-10:
Cool. For stuff like parts list, wiring, instructions - seems like a Creative Commons license would work. 

For reference, found like link about [Open Content](https://en.wikipedia.org/wiki/Open_content).

punamdahiya, 2016-05-26:
Opened issue #25 to further refactor and bring out copyright code in separate file. Closing this issue for this study.

closed 2016-05-26 by punamdahiya