mozilla/project_haiku.iot #21: Visualization for Logging

kglazko, 2016-05-09:
Create a better way of displaying data from the logs throughout the experiment. This can be a timeline with clusters of actions, both for a pair and for an individual. Doing this will allow staying informed and gather insights prior to the end of the experiment. This will also save time in terms of implementing data analysis post-experiment.

kglazko, 2016-05-10:
Notes from Liz: Some things we'd like to know

    For a button action, did recipient respond? Within a pair, was one person a more frequent sender than receiver?
    Assuming that there are some "clumps" of sends+receives, what does that look like (number sent/received, message type, etc)? By "clump" I mean a series of sends+receives in very close time. A "clump" would suggest that they were having a "conversation." That might be too specific though...

Some ways to visualize data

    By date/time - bar chart with each day having a bar composed of the number of sends and receives. This would also show us how sends compare to receives per person.
    By frequency - not sure how to visualize, but - a way to see patterns in their usage, e.g., does most activity occur during evening or during specific hours?

kglazko, 2016-05-10:
Will be hacking on this today

sfoster, 2016-07-07:
@kglazko can you link in the results here so we can close this issue?

oteo, 2016-07-11:
Data Analysis folder: https://drive.google.com/folderview?id=0B01m7pXAhdJmWXZlWnl1cGN1ODg&usp=drive_web.
I have included the redash and periscope dashboards links in each pair report included in the folder.

closed 2016-07-15 by sfoster