mozilla/project_haiku.iot #22: Document end-to-end instructions

ezoehunt, 2016-05-10:
Adding this as a reminder of what we discussed Mon afternoon:

1. Parts List with links to all parts. Include all specifics so if an item is out of stock at the link, reader can find the part elsewhere.

1.  Wiring diagram and instructions for wiring the parts up

1. 3D .stl file that describes the basic enclosure we've designed

1. Instructions that describe how to put together the parts/wiring with the enclosure.

1. Instructions for getting the code onto the device. May come in 2 flavors: 1) for devs who would use Github, and 2) for novices who would use a simplified process.

Once we've got that, I'll try to build the end-to-end device using these instructions and improve the instructions as I go. Then we publish!

punamdahiya, 2016-05-26:
This is dependent on Issue #18

punamdahiya, 2016-06-09:
End to End instructions documented here:
https://drive.google.com/a/mozilla.com/file/d/0ByJ_mHuB-es3ckNXVHVpNDFPTUU/view

ReadMe file updated with instructions:
https://github.com/mozilla/project_haiku.iot/tree/master/Prototype

sfoster, 2016-07-07:
@mermi was able to build out a device on a breadboard following the instructions. There's more we can do next time but for now, we are all done here

closed 2016-07-07 by sfoster