mozilla/project_haiku.iot #23: Log the time it takes to get a wifi connection and a ping

punamdahiya, 2016-05-16:
Log the time it takes to get the device connected to WiFi when it first comes online and respond to the click from paired device.

This is to investigate option to put the device to sleep (https://docs.particle.io/reference/firmware/photon/#sleep-sleep-) and improve battery life

kglazko, 2016-05-16:
`if (b.connecting()) {
	int startConnecting = Time.now();
}

if (b.ready()) {
	int successConnect = Time.now();
	int timeToConnect = successConnect - startConnecting;
	Particle.publish(CONNECTED_TO_WIFI, String(timeToConnect), 60, PRIVATE);
}`

kglazko, 2016-05-23:
Should we just put this in the set-up loop?

kglazko, 2016-05-23:
// We start with the setup function.
void setup() {
  Serial.begin(9600);
  b.begin();
  int startConnecting;
  if (WiFi.connecting()) {
    startConnecting = Time.now();
  }

if (WiFi.ready()) {
    int successConnect = Time.now();
    int timeToConnect = successConnect - startConnecting;
    Particle.publish(CONNECTED_TO_WIFI, String(timeToConnect), 60, PRIVATE);
  }
  b.attachClick(click);
  b.attachDoubleClick(doubleClick);
  b.attachLongPress(longPress);

  // Subscribe to paired device event event using Particle.subscribe
  // Subscribe will listen for the event SUBSCRIBE_EVENT_NAME and
  // when it finds it, will run the function handleEvent()
  Particle.subscribe(SUBSCRIBE_EVENT_NAME, handleEvent, MY_DEVICES);
};

sfoster, 2016-06-02:
If I understand correctly, this code isn't yet working properly. We want to capture the time it takes from reset and/or deep sleep to the time we are able to ping the particle service. If we can mark milestones along the way for acquiring a network connection etc. so much the better, but the key number here is how long will it be (could it be) to retrieve data. That includes startup time, time to acquire a network connection, get an IP address, resolve the particle.io name, issue a request and get a response.

punamdahiya, 2016-06-06:
To explore below option, first thing to find out when you power and log serial - Time needed

1. Power and WiFi connection - How long it takes
2. Power, start up and Particle Cloud - How long it takes
3. Difference between deep sleep and battery disconnected 
4. Based on the above time needed , we can determine if we can use battery powered WiFi in our devices

Based on the above data we can determine the feasibility of below idea: 
See if its possible to have device sleep up most of the time - ideally ping network every 5 seconds and make new connection and check with cloud service message queue if there is anything new

sfoster, 2016-06-06:
I started a spreadsheet with some initial results: https://docs.google.com/spreadsheets/d/19U1wRxKqDWiCAz9SEGUvcEZfHhzA5EnM32dGJ3jq4c8/edit?usp=sharing

sfoster, 2016-06-10:
So far its looking like we have a range from ~ 5s to 20s, which is a bit better than I had anticipated, but still slow enough that a power saving strategy to preserve the perception of real-time connectedness doesnt jump right out. It would be good to get more data to try to understand the range and distribution of these times we can expect. I'm assuming there will be outliers - networks are like that. 

Then, I think the next step is to write some code to test out sleep/deep sleep and see how those numbers compare.

sfoster, 2016-07-05:
With the data logged in https://docs.google.com/spreadsheets/d/19U1wRxKqDWiCAz9SEGUvcEZfHhzA5EnM32dGJ3jq4c8/edit?usp=sharing, this issue can be closed. Basically we are seeing a range from around 3 seconds from powering up to getting the response from the "ping", to about 20 seconds. We should file a new issue for the follow up work outlined in the previous comment.

closed 2016-07-05 by sfoster