mozilla/project_haiku.iot #27: Logging service that logs data from particle board

punamdahiya, 2016-05-26:
Create our own logging service to log data from device to an external file server / database

punamdahiya, 2016-06-06:
Need to figure out how to provision a server and an account for logging on cloud

punamdahiya, 2016-06-06:
Write a small node.js server that exposes REST API and  accepts post from specific URL and store data in database.

kglazko, 2016-06-07:
Discussed with Mahe and Sam and decided that GA/GBQ is a good solution for logging. I will be working on:

- Concreting out exactly what we want to log (what parameters/events/etc.)
- Deciding how we want to log (aka via direct POST in the code (bypassing particle publish), to GA directly or to Metrics API
- Getting some sample events to show up in Google BQ

mermi, 2016-06-29:
for the node.js server I have question the data base should be in a specific tool like mangodb ?!

sfoster, 2016-06-30:
@kglazko can you work with @mermi on what needs to happen here? As I understand it we can send data directly to the GA server. I'd like to see some prototype code using that API from the Particle button (i.e. in C++ running on device) so we can work out if it make sense to code to our own API and relay logging calls off to GA (or whatever backend), or make the connections directly from the device and avoid this middle-man node.js server altogether.

kglazko, 2016-06-30:
Here is an example of usage https://github.com/tamarahills/metrics_controller/blob/master/examples/node_demo.js

mermi, 2016-07-01:
yeah @sfoster I worked on that API with russ and Tamara, it is a rust library and now the library works with Java, c++ and JS client. So you wanna make the same thing ??

sfoster, 2016-07-01:
I want to get some working C++ demo code in our repo that demonstrates logging data from a Photon device using this API.

sfoster, 2016-07-07:
@mermi can you update this issue with the current plan. Perhaps we need to close this one and open a new ticket as it is mostly concerned with creating a logging service - which I believe it turns out we will not need.

mermi, 2016-07-15:
So today the c++ code works sending data from Particle to GA and here the code https://github.com/mermi/Logging-service/blob/master/logging.ino We are willing to get it works from Webhooks to GA now something happening that doesn't work from Webhook today will keep figure out the issue.

mermi, 2016-07-18:
concerning Webhook, I tried to use https://thingspeak.com for sending Data with Webhook to ThingSpeak and it works https://thingspeak.com/channels/135707 I just followed the documentation in Particle Docs We need to figure out why it doesn't work from Webhook to GA that time will look on it more with Kate when she is up to

closed 2016-08-29 by sfoster