mozilla/project_haiku.iot #31: Recycle existing hardware

corv89, 2016-06-06:
I've got an idea for a "smart home/IoT" type project I would like to implement.

Is it possible to re-purpose Firefox OS developer phones or a Raspberry Pi for Haiku?

sfoster, 2016-06-06:
We could prototype something with the old Flame phones, but there aren't enough out there for this to be a viable approach - we would need to figure out compatibility with at least a few more older android-capable phones. As we've seen with FxOS, that's quite a lot of work up-front, and ongoing. 

This project is steering towards a minimal-interaction, "ambient" communication device for teens, rather than anything "smart home" - which is why we changed the project name, so I'm not sure this is the best group to consider your idea. But if you want to join the #smarthome IRC channel we can discuss?

corv89, 2016-06-06:
Thanks for the heads-up, I'll stick to HTML5 then.  

Ambient communication sounds like a novel idea. I'll be keeping an eye on FirefoxOS's new directions.

Best of luck to your project!

closed 2016-07-07 by sfoster