mozilla/project_haiku.iot #37: HTML test client for status service

sfoster, 2016-07-15:
Lets have a test page that we can use as a prototype client for working out the interactions with the status server(s), and the lights, interactions etc.  

I outlined something in this gist...
https://gist.github.com/sfoster/da29b7f9242d955ca116586f6a5599f3

..but feel free to implement it however makes sense. Bear in mind this is throwaway code. It should be able to poll the server and light up the "LEDs" when it gets some status value it recognizes. Then we can work out the message format and other details here really quickly before porting to the device hardware and embedded code.

mermi, 2016-07-15:
Great, thank you Sam for the comments in the example and for the Canvas you would them display Rainbow lights ?!!

sfoster, 2016-07-15:
I'll get this started by adding a skeleton file to the status-server so we can load it directly from that server when it is running. 

@mermi sure. Lets treat each canvas as a Neopixel/LED so we can preview some different effects. You can then port over the rainbow function from the Arduino code

sfoster, 2016-07-29:
We have other issues covering next steps; this is complete as of 813e587c00bf87de6aaaa688b64703769e2bc7ee

closed 2016-07-29 by sfoster