mozilla/project_haiku.iot #53: Bill Of Material - BLE Buzzy Pixel

punamdahiya, 2016-07-22:
1. BLE Nano & MK20 USB Board - http://redbearlab.com/blenano/
2. [Optional] RedBearLab nRF51822 Pinout - http://redbearlab.com/redbearlab-nrf51822/ (Ideally BLE Nano should suffice, for beginners nRF51822 dev board is recommended instead of BLENano at first as it has easy pin out interface)
3. "Neopixel" WS2812 addressable LED strip. Non-waterproof, 60 LEDs per meter. (cut strip of 5 LEDs per unit) [Amazon](https://www.amazon.com/Mokungit-Programmable-Individually-Addressable-Non-waterproof/dp/B01D1EDDR8?ie=UTF8&*Version*=1&*entries*=0)
4. Sparkfun inventor kits part for resistor, buttons, led and jumper wires - [Sparkfun](https://www.sparkfun.com/products/13110)
5. Breadboard - [Sparkfun](https://www.sparkfun.com/products/12002)

sfoster, 2016-07-22:
Can you also drop this in a README alongside the code that goes with it?

punamdahiya, 2016-07-22:
Done. https://github.com/mozilla/project_haiku.iot/tree/master/Bluetooth/nRF51822/Arduino#bill-of-material---ble-peripheral-neopixel

closed 2016-07-22 by punamdahiya