mozilla/project_haiku.iot #65: SmartPhone App - Cordova Android app to handle communication even when app is closed

punamdahiya, 2016-08-01:
Scope of this issue is to investigate and implement Android app as background service that is it can
execute the code in the app even when app is closed.

punamdahiya, 2016-08-01:
Android has Services that could be used to keep connections alive while moving between activities, investigate if its possible to implement that using PhoneGap.

punamdahiya, 2016-08-01:
Related links:
http://stackoverflow.com/questions/10343828/creating-an-android-service-with-phonegap-have-phonegap-app-run-even-when-clos
http://www.codeproject.com/Questions/633535/Make-Phonegap-app-run-in-background

kglazko, 2016-08-11:
https://github.com/mozilla/project_haiku.iot/tree/backgroundtest Branch with perpetual background running on Android.

punamdahiya, 2016-08-16:
@kglazko quick question how do we test that the plugin let app live in background always? By default without plugin, I navigated to another app, forcing buzzy-pixel in bkg and see the app passing http requests.
Looks like thats Android default behavior, I am not sure if there is time cap to it and the background app is force killed after certain time.

punamdahiya, 2016-08-23:
Looks like there's a time cap, I have noticed after ~20 minutes of screen turned off, status doesn't update on connected BLE device.If screen is turend on after 20 minutes  request goes through and status updates. WIll integrate recommended plugin to test this use case.

punamdahiya, 2016-09-16:
Code branched to https://github.com/mozilla/project_haiku.iot/tree/Haiku_BLE_Beta1, closing as won'tfix

closed 2016-09-16 by punamdahiya