mozilla/project_haiku.iot #80: Test Client - Switch to using divs not canvas for LED/slots

sfoster, 2016-08-03:
Turns out we just need to animate color on each slot/LED, and any overlays or labels will be actually easier to accomplish with HTML/CSS.

mermi, 2016-08-03:
I already start that the CSS part, but I am going to make some change in the client.js because there in the init and render we are using canvas to change the color of each Led

sfoster, 2016-08-04:
@mermi in animations.html there's a setPixelNodeColor function that we could add to client.js. It supports both RGB and HSV colors which could be handy.

mermi, 2016-08-04:
So we would use it for instead of the Standard Rainbow one right @sfoster ??

sfoster, 2016-08-04:
Fixed and landed in 5946a9f5c4f1631763f515f4d6c3733a56d97f30, thanks!

closed 2016-08-04 by sfoster