mozilla/project_haiku.iot #93: SmartPhone App - Handle UI to connect/disconnect BLE device

punamdahiya, 2016-08-05:
Once a BLE device is configured and connected to a smart phone , it should remember the device and settings. It should stay connected unless explicitly disconnected via a disconnect button on the app.

Once a device is successfully setup and connected, subsequent load of the SmartPhone app should display the connected BLE device page with List view of friends and status.

If disconnected, BLE device should free up to be scanned and connected by another smartphone app.

atilag, 2016-08-08:
Android Bluetooth API offers some functionality to handle automatic reconnection, and "bonding" devices (so you don't need to pair them manually every time, just once). 
So a typical workflow to address the reconnection issue could be:
1. Start LE Scanning
2. When your device shows up in the list, you should connect to it [using](https://developer.android.com/reference/android/bluetooth/BluetoothDevice.html#connectGatt(android.content.Context,%20boolean,%20android.bluetooth.BluetoothGattCallback)): 
`device.connectGatt( context, true /* autoconnect */, connectCallback );` 
This will handle reconnection for you, so when the device you are connected to goes out of the scope, you will receive an "onConnectionStateChange" event into the connectCallback, warning you about the device going offline, but when it comes online again, this very same callback will warn you again about it too, so you can basically monitor the state of the connection, but don't need to call .connect() every time.

For the bonding part, there's a device.createBond() method call that will bond both devices, so the next time you perform a LE scanning, you will get the device instantly in the device list, and connect to it without pairing of any kind. Anyway, I haven't tried that with BLE devices before... so let me try a simple PoC program to see how actually behaves (the Android doc is very vague sometimes).

atilag, 2016-08-08:
As long as you are using PhoneGap, you may find this plugin better suited for your app https://github.com/randdusing/cordova-plugin-bluetoothle. It seems to have reconnecting and bonding features, so maybe supports encrypted communications as well.

punamdahiya, 2016-08-23:
Updated plugin, opened issue https://github.com/mozilla/project_haiku.iot/issues/154 for handling disconnect and reconnect to last connected device

closed 2016-08-23 by punamdahiya