mozilla/project_haiku.iot #105: Status Service - Do we need the `ok` field in the responses?

kdsolo, 2016-08-06:
Do we need the `ok` fields in the API responses? I think they're redundant because HTTP has status response:  `HTTP 200 OK`. In other cases, 400's-500's would be obvious non-successful requests and would not have an `ok` field either. 

For example, we're always replying with `true` here: 
https://github.com/mozilla/project_haiku.iot/blob/c794e4c3be37ac71bdcf1377560da4528fee191f/status-server/src/index.js#L42

sfoster, 2016-08-08:
Getting rid of the 'ok' sounds reasonable to me. Particularly in the status PUT cases where the request has updated the record - it makes sense there to just send the updated record as the response, so the client doesn't have to make a further request to see the new values.

kdsolo, 2016-08-08:
Great I will remove the `ok`s then.

kdsolo, 2016-08-09:
Forgot to close this from the PR

closed 2016-08-09 by kdsolo