mozilla/project_haiku.iot #114: UI Mock up for Mobile status

KatieCaldwell, 2016-08-10:


KatieCaldwell, 2016-08-12:
[image]
[image]

KatieCaldwell, 2016-08-12:
@sfoster @punamdahiya @ezoehunt here's a first pass at a simple wireframe (2 states) for status ONLY ui app. Feedback and thoughts requested.

KatieCaldwell, 2016-08-12:
question: when my status is unavailable, is this the same as OFF for this experiment? If yes, then like idea 1, when my status is OFF, my friends' status isn't available to me (all OFF).

sfoster, 2016-08-12:
Yes, my understanding for this initial experiment is that setting my own status to off/unavailable is effectively the same as going offline and turning off the device: all status changes and other updates are disabled. Maybe that isn't exactly what we want eventually, but this seems like a simple, testable assertion. If people dislike this, we'll hear why and I think that puts us in a better position than adding features at random

KatieCaldwell, 2016-08-12:
[image]

KatieCaldwell, 2016-08-12:
@sfoster v2. OFFline status

ezoehunt, 2016-08-15:
@KatieCaldwell  First ui mockups - yay!

In the toggle, do you press the black square to switch status?

I made a version to remove some words. What say you?

By "log out," do we mean log out of the entire application? For this version, we may need that. But I don't think we'd have "log out" on the physical wearable. Seems a part of the Bluetooth app..

[image]

sfoster, 2016-08-15:
This looks good to me. I'd like to file some issues to get this implemented. We'll make a start assuming its going to be something along these lines.

KatieCaldwell, 2016-08-16:
@ezoehunt 
Log out - was only because I was assuming (I know!) we'd need to log in somehow to get access the webpage and change your status.

And yes, the only interaction is toggling ON/OFF your own status. For this first pass, it's my understanding that this is the only requirement. 

Open Questions (for v2)
• How might we add/remove 'friends'

punamdahiya, 2016-09-16:
Code branched to https://github.com/mozilla/project_haiku.iot/tree/Haiku_BLE_Beta1, closing as won'tfix

closed 2016-09-16 by punamdahiya