mozilla/project_haiku.iot #117: Test Client - Mobile browser compatibility

sfoster, 2016-08-12:
We are going to want to build off the test/html client for our dogfooding/study mobile-web client. Lets start by ensuring we've got reasonable cross-browser compatibilty in the current major (mobile) browsers. We don't need exhaustive backwards compatibility, it should just work in the 1 or 2 popular/significant browsers for both iOS and android. I.e. Firefox for android, Safari (and Fx for iOS? not an iOS user, you tell me), Chrome. 

Off the top of my head: 
* Check and find alternatives to gecko-isms like Array.forEach. Do let and arrow functions work everywhere? May need to refactor if not. 
* CSS: I don't think we have any -moz- prefixes, but some unprefixed properties may need -webkit prefixed variations?
* Audio: our buzz sound clip is an OGG file - we'll need an MP3 as well. 

Please update this ticket with findings before starting any major changes so we can discuss.

mermi, 2016-08-16:
For the Audio there is a bunch of online converter we could convert the OGG file we have to MP3(this is already done) 
For CSS we will need to add the media queries for Specific Browser on iOS or Android with -webkit-min-device-pixel-ratio we will need to add -webkit for border radius and scale (I am already working on that in another css file named mobile.css) 
Searching for alternatives to Array.forEach didn't make any change on that yet 
now I have a quick question if I want to run the new html page on my phone how to do it ?!

mermi, 2016-08-16:
It seems liek the Array.forEach works well in needed browser but it takes some ms for that looking here: https://benhollis.net/blog/2009/12/13/investigating-javascript-array-iteration-performance/

sfoster, 2016-08-16:
AFAIK, the static methods on Array (vs the methods on the Array.prototype) were never implemented outside of Firefox/Gecko. Unless we observe a performance problem, I don't want to worry about micro-optimizations like these. And IIRC the results vary across browsers anyhow.

sfoster, 2016-08-16:
I just grepped *.js for Array.* and I don't see any instances of that pattern any more, so that's good. 
We do have some arrow functions though and a couple uses of `let`.

sfoster, 2016-08-18:
@mermi I took your PR and worked on it some: https://github.com/mozilla/project_haiku_status_api.iot/pull/6

sfoster, 2016-08-23:
The last of this landed in https://github.com/mozilla/project_haiku_status_api.iot/commit/e2c19db3fa18d78dc03abde73ea944ad0a021414

closed 2016-08-23 by sfoster