mozilla/project_haiku.iot #166: Make a plan for gathering metrics for September study

sfoster, 2016-08-29:
We'll want to be able to graph user activity  - status changes and messages. 
Lets identify what else needs to be logged, and where/how to implement this.

mermi, 2016-08-29:
I have discussed with russ and I knew now what we need to add and to do to integrate the metrics library to our Status API it's easy. But the thing we would discuss is the Key we need to record for Example we need to record the status of each user and the messages sent between them. 
and we could make the first scenario of our prototype as we have now : when we start first: all the users icons blink with yellow/green 2- toggle the button to on 3- the status will be changed and we have green light 4- send message (rainbow light) to another online user.

So that's my opinion for that I guess the thing we have to collect is status and messages and we can see the cycle (scenario) in the behavior of our prototype in the future visualizations.

sfoster, 2016-08-30:
That sounds about right to me. Ultimately, just like our earlier user study, we'll want to use the data to read the story. We'll want to be able to determine how often and how frequent users toggle status and send messages, and find patterns in use. We'll draw on the data to inform interview questions to help tease out how the prototypes fit into participant's day-to-day communications and interactions. 

@mermi go ahead and file a new issue to start integrating the metrics library into the status server. Please coordinate with @kdsolo if necessary as he knows this code best, and may have changes not yet merged into master. We'll also need to log data from the haiku app to provide context for the server-side logs. We can log charging and battery low events, bluetooth connections/disconnects, as well as possibly loss of network connection (assuming we can send those logs when a connection is next established.) We'll need an issue for that too.